whid-injector/WHID #1: "Is-anyone-around" function?

whid-injector, 2017-03-17:
Add, to the Atmega sketch, a function that will constantly check status changes of CAPSLOCK led and keep updated a red/green flag within each webpage in the ESP HTTP Server.
In order to detect if the victim is using the keyboard and potentially become aware of an ongoing attack.
In this way, the attacker (connected through wifi) will be constantly aware if someone is around the victim machine. This might increase the success rate and reduce the suspiciousness among victims.

P.S. Eventually it could even deployed with a preset automatic payload, that will run as soon as the WHID will not detect any victim's activity.

phosphore, 2017-05-30:
I like the idea of the capslock button, but in order not to arouse suspicions, it could be used something [like this](https://github.com/schollz/howmanypeoplearearound) to estimate the number of people in the vicinity using the approximate number of smartphones as a proxy, or even better (but maybe less applicable) measuring the WiFi received signal strength indicator (RSSI) like some people counter do. There's been a lot of research in recent years on this.
Keep up the good work :)

whid-injector, 2017-05-31:
Nice idea!
There is a little problem, meanwhile we might detect if a user is closer to the victim PC... we might also get other people around.
Now imagine if the victim PC is placed in a crowded place.... in this way we cannot be sure that the owner is really around or those detected phones belong to other people.

I understand that my method is less stealthy but has a better balance to narrow down if only the owner is around or not.

whid-injector, 2017-08-10:
Moving the issue to https://github.com/exploitagency/ESPloitV2/issues/4

closed 2017-08-10 by whid-injector