whid-injector/WHID #7: how to reset the Cactus?

jjjyyy01, 2017-05-19:
i have a  Cactus  to Install WHID Injector.i have some problems when i Install ESP_Programmer,it always become pro micro 3.3V  and can not install drive,then Arduino will say  programmer did not respond to command: exit bootloader.
so i want to reset the Circuit board，can u tell me the way to  reset the Cactus?﻿

jjjyyy01, 2017-05-19:
maybe i selected other board before,now board the red led no blinking

minkione, 2017-05-19:
http://wiki.aprbrother.com/wiki/Cactus_Micro_Rev2_Troubleshooting_and_FAQ#How_to_Revive_a_.E2.80.9CBricked.E2.80.9D_Cactus_Micro

minkione, 2017-05-19:
If it doesn't work... 99% is your computer.
- Try with another one (maybe on a USB 2.0 port)
- Change Arduino IDE version (sometimes it get screwed-up)
- Completely uninstall and reinstall the existing version of Arduino IDE

closed 2017-05-19 by whid-injector