whid-injector/WHID #9: Storage Full

MooonShot, 2017-06-10:
Hey I flashed with the Wifi Duck bin file after watching your video, I successfully flashed that but When I connect to AP I can't upload scripts It says storage full and Format button is not working

[image]

whid-injector, 2017-06-10:
You made some mistake for sure.
And also check how to open tickets!!!
Read ot all carefully!
https://github.com/whid-injector/WHID/wiki

closed 2017-06-10 by whid-injector

whid-injector, 2017-06-10:
This issue shows that the FS in the ESP didn't get initialized/formatted properly.
You have to follow exactly how is showed in the video. With the same settings.
If still persists might be your Arduino Version or the fact you use a USB 3.0 port.

About the reset Hall sensor it works. You just need to get used to it. (And of course, u need a magnet)

MooonShot, 2017-06-10:
I did everything stated in video, I have USB 2.0 and It's not letting me upload payload with your firmware. This thing does not work. I need refund!

whid-injector, 2017-06-10:
1) I am NOT the seller and I have ZERO profit on it.
2) Is like 99.99999% impossible is a defective piece. 
3) I have tested 20+ devices and so far NO ONE broken. At most I had to reset it.
4) If people is not skilled enough to understand how one opensource  gadget works... better don't buy and waste time of other people. 🤝

So far you are the first I see with this issue.

MooonShot, 2017-06-10:
I very well know how gadget works, Any way to replace my product ?

whid-injector, 2017-06-10:
Read point 1) and 4) again.

MooonShot, 2017-06-10:
_/\_

whid-injector, 2017-06-10:
Try contact the seller and proove them the HW is faulty.