whid-injector/WHID #10: SPIFFS filesystem underutilized

exploitagency, 2017-06-26:
In the video you suggest setting up a generic esp8266 board in the Arduino IDE with 512kb module and a 64kb file system IIRC.

In reality WHID Cactus and the original ESPloit use an ESP12E chip with 4M flash and up to a 3M file system.  So your underutilizing a ton of space.

I just needed to throw this out there in case someone had an unreasonably large set of payloads and had issues there.

minkione, 2017-06-27:
Of course it depends on the HW.
For cactus WHID the ESP12 can fit 4M/3M.
In some Videos is even showed. ;)

minkione, 2017-06-27:
Like this one for example https://youtu.be/PGa_ByyQw8Q

minkione, 2017-06-27:
I guess you are referring to this video https://www.youtube.com/watch?v=MRGUSPW-Cr0&sns=em

It was made with the Cactus Micro r2.
That's why in the title is not mentioned the Cactus WHID Hardware.

Generally I assume that whoever is going to play with WHID has basic understanding of Arduino and ESP versions. :)

But thanks for the remark! 🤝😉

exploitagency, 2017-06-27:
Roger that!

closed 2017-06-27 by exploitagency