whid-injector/WHID #13: ESP_Programmer Error

sheepsleg777, 2017-08-07:
I will admit that I am noob to Arduino. I have been reading and trying to program this Cactus WHID device for a month now. Just following all of the various links is enough to make on want to eat a bullet. I have started to compile my own "how to guide" from all the various information out there. I do not take credit for any of it.  I keep getting hung up at the ESP_Programmer sketch loading. I always get an error with no printout on what it is. I have verified the code and all seems well. I have tried two separate devices with no luck. I have lowered the baud rate to 9600, no luck. I have remade the ESP_Programmer, no luck. I have completely reinstalled the Arduino IDE (v1.8.3), no luck. I am currently using my windows 7 64 bit home machine on a USB 2.0 port. Device manager sees Lilypad Arduino USB (COM3).  The youtube videos I have also reviewed. I am just trying to get over the ESP_Programmer hump. Any help or ideas would be much appreciated. Below is the hot to guide up to the point I have reached. Thanks

WINDOWS - HOW TO CONFIURE (WIFI DUCKY / WHID ) WITH ARDUINO IDE (V1.8.3) & ESP8266 FLASHER
---------------------------------------------------------

1) DOWNLOAD AND INSTALL CURRENT: Arduino IDE (Version 1.8.X) & NODEMCU-FLASHER TOOL

2) CONFIGURE ARDUINO IDE

*ADD BOARD MANAGER URL:
	1- OPEN ARDUINO IDE > FILE > PREFERENCES 
	2- ADD "http://arduino.esp8266.com/stable/package_esp8266com_index.json" TO BOARD MANAGER URL CLICK OK
	
*INSTALL BOARD MANAGER:
	1- TOOLS > BOARD > BOARDS MANAGER > TYPE = ALL FILTER = ESP8266 (VERSION 2.3.0) > INSTALL
	2- RESTART THE ARDUINO IDE PROGRAM (MAY NOT BE NECESSARY, BUT WAS ADVISED TO)

*INSTALL THE ESP8266 LIBRARIES: 
	1- DOWNLOAD ESPAsyncTCP & ESPAsyncWebServer ZIP FILES (UNZIP AND REMOVE MASTER AND UNDERSCORES FROM FOLDER AND REZIP THEM. THEY SHOULD LOOK LIKE THIS WHEN 	   DONE, AND IF YOU WERE TO DRILL INTO THEM "ESPASYNTCP.ZIP & ESPASYNCWEBSERVER.ZIP"
	2- OPEN ARDUINO IDE
	3- SKETCH TAB > INCLUDE LIBRARY > ADD.ZIP LIBRARY > SELECT ONE OF THE ABOVE FILES AND REPEAT PROCESS FOR SECOND ONE
	4- VERIFY THEY SHOW UP IN AVAILABLE LIBRARIES: SKETCH TAB > INCLUDE LIBRARY > SCROLL UNTIL YOU SEE BOTH FILES

3) COPY CODE AND BUILD .INO FILES THAT ARE NEEDED > OPEN ARDUINO IDE > NEW > DELETE FIELD THEN PAST CODE > SAVE AS REPSECTED FILE NAME. 

	1- ESP_PROGRAMMER
	2- ARDUINO_WIFI_DUCK
	3- ESP8266_WIFI_DUCK
	4- ARDUINO_SKETCH
	5- ESP_SKETCH

4) VERIFY WINDOWS SEES DEVICE IN DEVICE MANAGER (USE USB 2.0 NOT 3.0 PORT) 
	1- DEVICE MANAGER > PORTS > EXPAND TREE > ARDUINO LILYPAD USB (COMX) MINE SAYS COM3

5) SET ARDUINO IDE TO SEE DEVICE
	1- TOOLS > BOARD > LILYPAD ARDUINO USB
	2- TOOLS > PORT > COM3 (LILYPAD ARDUINO USB)

6) OPEN ESP_PROGRAMMER.INO IN ARDUINO IDE
	1- VERIFY (THE CHECK MARK AT TOP LEFT OF SCREEN) IF NO ERRORS GOTO STEP 2 IF ERRORS CHECK THE CODE FOR ERRORS
	2- UPLOAD (IT MAY BE SLOW BUT JUST LET IT RUN ITS COURSE)

sheepsleg777, 2017-08-07:
Update on specific error,
Enabling verbose output during upload revealed the follow:

Arduino: 1.8.3 (Windows 7), Board: "LilyPad Arduino USB"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\root\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\root\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\root\Documents\Arduino\libraries -fqbn=arduino:avr:LilyPadUSB -vid-pid=0X1B4F_0X9208 -ide-version=10803 -build-path C:\Users\root\AppData\Local\Temp\arduino_build_146223 -warnings=none -build-cache C:\Users\root\AppData\Local\Temp\arduino_cache_474627 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\root\Documents\Arduino\ESP_Programmer\ESP_Programmer.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\root\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\root\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\root\Documents\Arduino\libraries -fqbn=arduino:avr:LilyPadUSB -vid-pid=0X1B4F_0X9208 -ide-version=10803 -build-path C:\Users\root\AppData\Local\Temp\arduino_build_146223 -warnings=none -build-cache C:\Users\root\AppData\Local\Temp\arduino_cache_474627 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\root\Documents\Arduino\ESP_Programmer\ESP_Programmer.ino
Using board 'LilyPadUSB' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=8000000L -DARDUINO=10803 -DARDUINO_AVR_LILYPAD_USB -DARDUINO_ARCH_AVR  -DUSB_VID=0x1B4F -DUSB_PID=0x9208 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="LilyPad USB"' "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\leonardo" "C:\Users\root\AppData\Local\Temp\arduino_build_146223\sketch\ESP_Programmer.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=8000000L -DARDUINO=10803 -DARDUINO_AVR_LILYPAD_USB -DARDUINO_ARCH_AVR  -DUSB_VID=0x1B4F -DUSB_PID=0x9208 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="LilyPad USB"' "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\leonardo" "C:\Users\root\AppData\Local\Temp\arduino_build_146223\sketch\ESP_Programmer.ino.cpp" -o "C:\Users\root\AppData\Local\Temp\arduino_build_146223\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files (x86)\Arduino\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\root\AppData\Local\Temp\arduino_build_146223\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega32u4 -DF_CPU=8000000L -DARDUINO=10803 -DARDUINO_AVR_LILYPAD_USB -DARDUINO_ARCH_AVR  -DUSB_VID=0x1B4F -DUSB_PID=0x9208 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="LilyPad USB"' "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\leonardo" "C:\Users\root\AppData\Local\Temp\arduino_build_146223\sketch\ESP_Programmer.ino.cpp" -o "C:\Users\root\AppData\Local\Temp\arduino_build_146223\sketch\ESP_Programmer.ino.cpp.o"
Compiling libraries...
Compiling core...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc" -c -g -x assembler-with-cpp -flto -MMD -mmcu=atmega32u4 -DF_CPU=8000000L -DARDUINO=10803 -DARDUINO_AVR_LILYPAD_USB -DARDUINO_ARCH_AVR  -DUSB_VID=0x1B4F -DUSB_PID=0x9208 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="LilyPad USB"' "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\leonardo" "C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\wiring_pulse.S" -o "C:\Users\root\AppData\Local\Temp\arduino_build_146223\core\wiring_pulse.S.o"
[... cut ...]
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : butterfly
         Description     : Atmel AppNote AVR109 Boot Loader

Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding

avrdude: butterfly_recv(): programmer is not responding
Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

whid-injector, 2017-08-10:
Try to fully uninstall the Arduino IDE 1.8.x (included the junk into %AppData%). Reboot PC. Download & installl the 1.6.x version.
I also assume that you tried already to reset WHID with the magnet as described in the wiki?!
Did you try with WHID SW? And also with WifiDucky? And ESPploitV2? All of those sketches fail?

sheepsleg777, 2017-08-11:
Every sketch / flash / esptool.py fails on multiple machines. This is to include a linux box as well, using IDE and pytool.py. The reason is always because the device is not responding even though it is seen by all of the computers that I have used. I have downgraded to 1.6.5 IDE and get the same error. Attempting to do the hard reset with a magnet reveals no hidden ports as described by this wiki. The board does go into reset mode and device manager cant see it, it comes back online about 8 seconds later. IDE only shows the the com3 port available at all times. Attempting to upload anything during the reset process and IDE says it can find the device until it comes out of reset mode. Then we are back to not responding error.  If no other ports show up, the only choice I have is the com3. When you is you command AT in serial monitor it hangs up as if its waiting for it to respond. I have tried wifiducky / esploitv2, but am unsure what the whid sw is? Got any more ideas? Maybe something got fried going through customs or maybe an aircraft radar zapped something? Thanks for any info.

volca, 2017-08-11:
For upload sketch for arduino (LilyPad Arduino USB), I think pytool.py or esptool.py is not needed. Do you have any other laptop to test the Cactus? 

> When you issue the command AT in serial monitor it hangs up as if its waiting for it to respond.

The esp8266 is not flashed with AT firmware. AT command is not worked.

sheepsleg777, 2017-08-12:
Solved! The solution in the end was a really old msi u120 notebook with bt5r1 with arduino ide v1.8.4. It was the fifth computer that I tried. Crazy. Thanks all.

closed 2017-08-12 by sheepsleg777