whid-injector/WHID #14: About Us

hex4d0r, 2017-09-09:


hex4d0r, 2017-09-09:
[https://i.hizliresim.com/pG4oqJ.png](img)

There is a problem about printing characters. I didn't tried others only Turkish. I know there's no problem about tr_TR lang file because i made it. I think It's about Json decode(or encode) or a HTML problem.

exploitagency, 2017-09-10:
https://streamable.com/ve5yh
Please post more information, post the payload you were attempting to run, the current version of the software that you are running, and your source for the Keyboard.cpp file.

hex4d0r, 2017-09-10:
It happens when i try to print special characters in my language. [ıİçşğ] etc.

exploitagency, 2017-09-10:
There seems to be an issue with live payload mode and input mode, when it comes to these characters.

Try writing a payload then uploading that payload and running it.  The characters won't be displayed properly when you view the payload in the web view, instead click on the download payload and you can see that the payload uploaded correctly.

Let me know how this goes for you when you run the payload that you uploaded.

Again do not use live payload mode or input mode, test with upload payload and choose payload then run payload and view payload using the download button.

exploitagency, 2017-09-10:
Try uploading the payload and running it, do not use live payload mode or input mode.  Let me know how it goes and report back with the requested info.

whid-injector, 2017-10-26:
Any news @hex4d0r?
Can I close the issue?

closed 2017-11-04 by whid-injector