whid-injector/WHID #15: HID MITM

GarethWright, 2017-09-12:
What is the latitude for using this to MITM a HID device for remote keylogging etc?

minkione, 2017-09-14:
This HW has different purpose from MiTM HID devices.

GarethWright, 2017-09-14:
I realise this but I've yet to see a MiTM HID with ex filtration by WiFi

minkione, 2017-09-14:
Look better then. http://www.keelog.com/wifi_hardware_keylogger.html

GarethWright, 2017-09-14:
Funny, though perhaps I should have been more specific

- with it's own WiFi access point
- doesn't cost stupid money

Do you not think that there's latitude for a device with both MiTM and Ducky capability. Particularly as the ESP8266 is so cost effective?

whid-injector, 2017-09-18:
you could, but is not within the scope of WHID device act as hardware MiTM.
For that exists Wifi Keyloggers.

closed 2017-09-18 by whid-injector

br0k3nilluzion, 2017-09-18:
there are aleady examples of this being done
https://github.com/spacehuhn/wifi_ducky/issues/40#issuecomment-327694511

https://github.com/supersjimmie/wifi_ducky_keylogger

GarethWright, 2017-09-18:
Thanks @br0k3nilluzion, exactly what I was looking for

exploitagency, 2017-09-18:
You need additional hardware (usb host shield).

GarethWright, 2017-09-18:
Yeah, that's not an issue. Just about getting something hardware based, and small enough to be able to demo an issue to the higher ups, who seem to think AV's are the be all and end all.