whid-injector/WHID #19: Update ESPloit Submodule to latest commit

exploitagency, 2017-10-24:
Submodule is on an old commit.  Please consider pulling latest commit for ESPloit from my repo.

I have added a lot of new features to ESPloit and even made some minor changes to the 32u4 code.

whid-injector, 2017-10-26:
Done!
P.S. Due to all changes I guess that the FW upgrade on-the-fly cannot be achieved (since within the last release, it is also mandatory to upload the 32u4 sketch). Right?

closed 2017-10-26 by whid-injector

exploitagency, 2017-10-26:
No it will still work, it will just have limited features and tell you to manually update the 32u4 sketch when you click on upgrade firmware.

minkione, 2017-10-27:
Perfect! Just wanted to be sure people won't brick WHID with improper flashing the FW.

whid-injector, 2017-10-27:
Perfect! Just wanted to be sure people won't brick WHID with improper flashing the FW.