whid-injector/WHID #21: Brackets

20giro, 2018-01-14:
Hi, i'll probably be asking a trivial question, but how can i write sqare brackets and braces ?
When i try to put them in a payload, different characters are written.
Thanks for the help.

20giro, 2018-01-20:
I mean, how can I write all the characters that need combinations with altgr and shift?

whid-injector, 2018-01-20:
1) which software you using? ESPloitV2? If yes, open ticket in that repo. https://github.com/exploitagency/ESPloitV2

2) Special Characters/Known Issues:
Currently the only character that has been found not to work is the "less than" symbol, "<".
This bug does NOT apply to Live Payload Mode.
This only applies if you are using Upload Payload, the script will stop uploading when it reaches the "<" symbol.
The work around for writing a script that requires a "<" is to replace all instances of "<" with "&lt;".
The script will upload properly and when viewed and/or ran it will replace "&lt;" with "<".

whid-injector, 2018-01-20:
Moreover... which Language is your target? US?
If not, change the keyboard language of arduino IDE accordingly.

whid-injector, 2018-01-20:
And read all wiki!

closed 2018-01-20 by whid-injector

20giro, 2018-01-20:
Yes, I'm using ESPloitV2, I'm going to open a new ticket there. 
I have already red the wiki, I'm using Italian layout and I have already changed the Arduino language. For example I have a problem  when I try to write "{" is written "è".
Thanks for the help.

whid-injector, 2018-01-20:
Paste here the payload you are trying to execute.

I will try to check it (when I'll have time next week)

20giro, 2018-01-20:
I'm not talking about any payload, I was trying to write some shells to try the cactus and I found out this problem.
If I just type "{" is written "è", it's like the cactus can't type the combinations with shift, alt, altgr and keys like these.

whid-injector, 2018-01-20:
I need a payload to troubleshoot.

whid-injector, 2018-01-20:
Moreover... which OS and if it is an Apple product

20giro, 2018-01-20:
I'm on Windows 10
The payload could be:

PrintLine: {

I know this is a useless payload but I can't understand why instead of the brace it wrote "è"

whid-injector, 2018-01-20:
This is an Arduino and Keyboard layout issue then.
I would suggest to open a ticket on Nurrl's LocaleKeyboard.js github repo.

Also check arduino forum.For sure someone already had similar issues.

Sadly I have no italian keyboard. And changing layout on OS and use a KB with US layout will make troubleshooting a mess.