whid-injector/WHID #22: Recovery

hanshaze, 2018-04-04:
My WHID Injector was formatted.....and i have no idea wtf i have to do to get this nice little tool back to life again......wish it would be easy as the rubberducky fw-flash was back in the days....so heres the question --->
how can i restore to the default Settings to be able to use all functions?

Please explain how to do with Arduino IDE and the FlashingTool-exe....

Do i have to flash fw twice? if yes, go further into details please......

Why are in this FlashingTool under the Tab "Configuration"  a few empty boxes for bin-files to use?
Do i have to add more than one bin? If yes, please tell me which one i have to choose?

And whats the point of the adresses of the bin files e.g. 0x0000, 0x01000 and so one? 
I think it starts with 0x0000 and goes upon with every bin file, right?

exploitagency, 2018-04-04:
Formatted?  Formatting SPIFFS through the web interface is safe and if cancelled before completion simply restores the default config, if you wait it saves current config.  I hope this is what you mean.

Otherwise follow the flashing instructions and videos.

exploitagency, 2018-04-04:
Also here are serial recovery instructions if you have not literally flashed your device.

https://github.com/exploitagency/ESPloitV2

hanshaze, 2018-04-05:
ok......my mistake was to select the Connect-to-my-WLAN-option via web-interface without setting up creds.....after reboot the stick wasnt available via browser......and  200 "flashed" ino files later im here......at least, after i followed the readme the ide gave me something like "enjoy it" as message in the console.....but still no connection to 192.168.1.1........hope you can help me......if not.....well, nice looking usb-stick-dummy :-D

exploitagency, 2018-04-05:
You could have sent a simple command over serial to recover in that situation.  Should have read the docs...

Now you have to follow the instructions for flashing.

exploitagency, 2018-04-05:
https://github.com/exploitagency/ESPloitV2/blob/master/README.md#resetting-to-default-configurationrecovering-device

exploitagency, 2018-04-05:
Either way it will be easy to recover.  Please be sure to update us.

whid-injector, 2018-04-06:
[image]

whid-injector, 2018-04-06:
Looong time ago I also recorded a video... which is mentioned in the wiki you probably didn't read (carefully)... https://github.com/whid-injector/WHID/wiki#how-to-configure-whid-software-windows

https://www.youtube.com/watch?v=MRGUSPW-Cr0

closed 2018-04-06 by whid-injector

hanshaze, 2018-04-06:
Words of wisdom....
Thanks for help anyway ^^

whid-injector, 2018-04-06:
Sorry if I sounded too sarcastic!
It is friday! 
[image]

hanshaze, 2018-04-06:
ok.....stopped 5mins ago tweaking for today............

heres what ive done so far

- doubleclick the first ino file ( WHID --> Tools --> ESP Programmer)
- add JSON package...whatever
-build and upload to the dongle
[image]

- unplug dongle, close all windows
- use terminal to type in this command --> "sudo python esptool.py -p COMxx erase_flash"

- now i doubleclicked the 2nd fw and compiled it to a bin file, exported it and closed the arduino ide
- next step was to open the flashtool.exe as admin, browse to the bin file, checked the other settings like     port and stuff like that,
[image]

- finally just push the button #plugnpray
[image]

[image]

hanshaze, 2018-04-06:
[image]

[image]

exploitagency, 2018-04-06:
Did you even try my suggestion?  

Why are you trying to flash 3 different firmware onto the ESP?

BE SURE TO FLASH BACK THE 32U4 SKETH BEFORE TESTING THE ESP!

The 32u4 sketch literally turns on the esp enable pin.

exploitagency, 2018-04-06:
There are precompiled binaries in the ESPloit repo I linked.  I suggest you use the latest one from the releases page.

exploitagency, 2018-04-06:
Luca can you add the serial recovery instructions I linked earlier to somewhere obvious in your wiki page in case someone locks themselves out of the web interface again.  This should have been a 5 second fix and not a flashing tutorial.

hanshaze, 2018-04-06:
I stop now with the shit, would be more productive to sort logfiles .... have to take an hour or two to count sheep ^ ^

hanshaze, 2018-04-10:
If someone is interested, thats a quick description ---->

Step 1:
Open Arduino IDE, Insert dongle, Open ESploit esp8826-programmer-ino, compile and upload to dongle

Step 2:
Open ESP-Sketch-ino, compile to bin via IDE and upload somehow (flasher, esptool, ...) to dongle

Step 3:
Open 32u4-ino via IDE, compile and upload.......

Step 4:
Check for WLAN "Exploit" with  pw "DotAgency"

Where do i edit the password in the ino file?
Because i think i have to edit before i compile......

whid-injector, 2018-04-10:
in ESPLOITV2 you can change in the settings. No need to modify any .ino...

Read esploitv2 readme!

Best Regards,
Luca

exploitagency, 2018-04-10:
Stop ignoring my advice...  I wrote ESPloitv1/v2.  Read the below and issue `ResetDefaultConfig:OpenNetwork` via serial.

Resetting to default configuration/Recovering device
Plug the device into your computer.
Open the Arduino IDE.
Select Tools - Board - "LilyPad Arduino USB".
Select Tools - Port and the port the device is connected to.
Select Tools - "Serial Monitor".
Select "38400 baud".
Now type in "ResetDefaultConfig:". Without the quotes but be sure to include the colon symbol.
Click Send.
You should now receive the following reply "Resetting configuration files back to default settings."
Wait about 15 seconds or until the LED blinks and unplug and replug in the device.
The device has now been reset back to default settings.
Connect to the Access Point with the SSID "Exploit" with a password of "DotAgency".
Open a web browser pointed to "http://192.168.1.1"
The default administration username is "admin" and password "hacktheplanet".

NOTE: Certain devices seem to have trouble connecting to a password protected ESP8266 access point. The symptoms of this involve repeatedly being prompted to enter the password and being unable to connect to the ESP8266 via WiFi. This can be solved by following the above instructions but instead issuing the command "ResetDefaultConfig:OpenNetwork" via serial. The device will be restored to the factory defaults(with the exception of now being an unsecured network). The device will reboot and you may now connect to it as an unsecured WiFi access point with an SSID of "Exploit". You should now be able to establish a connection.

exploitagency, 2018-04-10:
Now the latest ESP SDK does have some bug fixes and I plan on making sure everything is backwards compatible soon with the latest SDK and then also perhaps trying to run the ESP at 160MHz but that will take some time to test as ESPloit has a ton of different features and advanced settings.  But when you get more comfortable with things perhaps you should give it a shot as well.

hanshaze, 2018-04-12:
man, how stupid I have been ..... 3 days of my life wasted for a matter of not even a minute :-D knew how to save well known energy ..... lol

exploitagency, 2018-04-12:
You finally did the serial recovery and now your problem is solved right?

hanshaze, 2018-04-12:
yes, but somehow the last few times i became no output back in the serial mon console......maybe the reason for this is somehow connected to the devices in my other ports.......and i dont mean wrong port conf......but anyway, now i could start a timeattack challenge in reflashing ^^

hanshaze, 2018-04-22:
ok ^^