whid-injector/WHID #24: Add Karma Functionality to ESPortal Credential Harvester

whid-injector, 2018-06-06:
Just found this and looks promising to integrate in the ESPortal Ceredential Harvester.
https://advancedpersistentjest.com/2018/06/03/karma-on-esp8266/amp/
https://github.com/CreateRemoteThread/esp8266-karma
https://github.com/pulkin/esp8266-injection-example

exploitagency, 2018-06-06:
I believe I would have to downgrade the ESP SDK just by glancing over the docs.  Not sure what would break.  The new SDK is actually pretty good now compared to when I started.  I even tested some overclocking to 160 and it seemed fairly stable on new sdk.

I will look into it eventually though.  Thanks for sharing Luca!  I have seen a little bit related to this.

I have a lot going on offline(IRL) right now but will be getting back to writing code eventually.  I mainly have been afk keeping tabs on things from my mobile lately.

So, I started treating GitHub too much like a social networking site and then Microsoft went and bought it...

Anyways lots of ideas. I just need about a year long vacation to catch up on all of my projects. I have boards for various prototypes I designed and components scattered everywhere, not to mention lots of untested code I wrote for said devices scattered across various drives.

Anyways will keep you in the loop.

ChandraOrbit, 2018-10-07:
When is it..?