whid-injector/WHID #26: WHID damaged ?

michelep, 2018-09-13:
Hi,

i have some troubles with my WHID and i can't do anything and still don't work. When i plug it into USB port i get:

Sep 13 15:29:50 usb 3-8: new full-speed USB device number 21 using xhci_hcd
Sep 13 15:29:50 usb 3-8: New USB device found, idVendor=1b4f, idProduct=9203
Sep 13 15:29:50 usb 3-8: New USB device strings: Mfr=2, Product=1, SerialNumber=0
Sep 13 15:29:50 usb 3-8: Product: Pro Micro 3.3V
Sep 13 15:29:50  usb 3-8: Manufacturer: SparkFun Electronics
Sep 13 15:29:50 cdc_acm 3-8:1.0: ttyACM0: USB ACM device
Sep 13 15:29:50 mtp-probe: checking bus 3, device 21: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-8"
Sep 13 15:29:50 mtp-probe: bus: 3, device: 21 was not an MTP device
Sep 13 15:29:52 usb 3-8: USB disconnect, device number 21

As you can see, device disconnect after two seconds..:-(

Any hint ?

whid-injector, 2018-09-13:
What lsusb says?!
After 5 mins do u see the wifi SSID?
Were you flashing it before it happened?
Was working before and then stopped?

P.s. Read also these points I need to know!
https://github.com/whid-injector/WHID/wiki#how-to-open-a-ticket

The more info, the better.

minkione, 2018-09-13:
I believe that is an issue with MTP... did u try on another PC?
https://ubuntuforums.org/showthread.php?t=2057988

whid-injector, 2018-09-28:
Any news? 
Solved?
Can I close the issue?

michelep, 2018-09-29:
I'm still waiting for the programmer to try fixing ATmega chip...

michelep, 2021-05-23:
I have spend some time trying to fix the issue... AVRdude tell me:

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/michele/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf"
         User configuration file is "/home/michele/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyACM0
         Using Programmer              : avr109
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega32U4
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : butterfly
         Description     : Atmel AppNote AVR109 Boot Loader

Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude: devcode selected: 0x44
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude: reading input file "/tmp/arduino_build_230955/esp8266Programmer.ino.hex"
avrdude: writing flash (4960 bytes):

Writing | ##########avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: set addr
#####avrdude: error: programmer did not respond to command: write block
# ***failed;  
 ***failed;  
 ***failed;  
[..] 
avrdude: Error: butterfly programmer uses avr_write_page() but does not
provide a cmd() method.
 *** page 95 (addresses 0x12e0 - 0x135f) failed to write

# | 100% 12.05s

avrdude: 4960 bytes of flash written
avrdude: verifying flash memory against /tmp/arduino_build_230955/esp8266Programmer.ino.hex:
avrdude: load data flash data from input file /tmp/arduino_build_230955/esp8266Programmer.ino.hex:
avrdude: input file /tmp/arduino_build_230955/esp8266Programmer.ino.hex contains 4960 bytes
avrdude: reading on-chip flash data:

Reading |

and stay stuck here. I think AVR just say good bye :-(