whid-injector/WHID #27: Change keyboard layout

PaulSec, 2018-09-17:
Hi Luca, 

I tried changing the keyboard layout of my WHID to fit french keyboards. 
My current OS is : ```Linux cert1 4.14.0-kali3-amd64```.  

I personally: 
1. changed the Keyboard.cpp constant values, 
2. Plugged the WHID on USB 2 port
3. Opened the Arduino Sketch in this repository

```c
/*
 * WHID Injector - Arduino Sketch
 * http://whid.ninja
 * Forked by Luca Bongiorni
 * Based on ESPloit by Corey Harding of www.LegacySecurityGroup.com
*/

/*
For individual keypresses or combinations of key presses send the following via software serial:
-"Press:X" or "Press:X+Y" or "Press:X+Y+Z" and so forth
--Expects DECIMAL Values for X,Y,Z,etc
--Sending the following via software serial "Press:131+114" with the Arduino's USB plugged into a Windows machine would output KEY_LEFT_GUI(Windows_Key)+r thus launching the run prompt
--List of modifier keys(GUI,ALT,CTRL,ETC) at https://www.arduino.cc/en/Reference/KeyboardModifiers
--ASCII table lookup at http://www.asciitable.com/

To type out strings of text send via software serial:
-"Print:XYZ" Types out "XYZ"
--Sending the following via software serial "Print:www.Exploit.Agency" would type out "www.Exploit.Agency" on the machine connected via USB

To type out strings of text send via software serial:
-"PrintLine:XYZ" Types out "XYZ" then presses enter
--Sending the following via software serial "PrintLine:www.Exploit.Agency" would type out "www.Exploit.Agency" on the machine connected via USB and press enter

To make a delay:
-"Delay"
--Sending the following via software serial "Delay" would wait for X seconds(length of delay set in esp8266 sketch) on the machine connected via USB before proceeding to next item in payload
*/

#include <Keyboard.h>
#include <SoftwareSerial.h>
//#include <HID-Project.h>
//#include <HID-Settings.h>

//Used later for determining if we are ready to release a key press or a combination of key presses
int keypressdone=0;

void setup() { 
  pinMode(13, OUTPUT);
  digitalWrite(13,HIGH);
  Serial1.begin(4800);
  Keyboard.begin();
}

void loop() {  
  while (Serial1.available()) {
    String cmd = Serial1.readStringUntil(':');
    //If command equals "Press:X" or "Press:X+Y+ETC"
    if(cmd == "Press"){
      keypressdone=1;
      String fullkeys = Serial1.readString();
      int str_len = fullkeys.length()+1; 
      char keyarray[str_len];
      fullkeys.toCharArray(keyarray, str_len);
      char delimiter[] = "+";
      char *keypart;
      keypart = strtok(keyarray, delimiter);
      while(keypart != NULL) {
        int key = atoi(keypart);
        delay(25);
        Keyboard.press(key);
        keypart = strtok(NULL, delimiter);
      }
      
      if(keypressdone==1) {
        delay(25);
        Keyboard.releaseAll();
        keypressdone=0;
      }
    }
  
    //If command equals "Print:X"
    else if(cmd == "Print") {
      String keycode = Serial1.readString();
      delay(25);
      Keyboard.print(keycode);
    }

    //If command equals "PrintLine:X"
    else if(cmd == "PrintLine") {
      String keycode = Serial1.readString();
      delay(25);
      Keyboard.print(keycode);
      delay(25);
      Keyboard.press(KEY_RETURN);
      delay(25);
      Keyboard.release(KEY_RETURN);
    }
  }
}
```

3. Selected Lilypad Arduino USB for the board
4. Chose port /dev/ttyACM0 
5. Clicked Upload 

I got: 

```bash
Sketch uses 8922 bytes (31%) of program storage space. Maximum is 28672 bytes.
Global variables use 437 bytes (17%) of dynamic memory, leaving 2123 bytes for local variables. Maximum is 2560 bytes.
```
The sketch seems to have been uploaded successfully, I manage to connect to the Wi-Fi network (unchanged) but when I try launching commands using the Input Mode, nothing happens on the host machine. 

Let me know if you need further details !

whid-injector, 2018-09-17:
So far all the process looks fine.
Did u cross-check with the video in wiki if u followed exactly all steps?

The usb port was 2.0, right?

Which Arduino IDE version is?

Did u try a softreset over serial or hardreset with magnet and reflash the sketch? (check wiki amd youtube channel)

@exploitagency maybe you have other questions?
I cannot think anything else that could have been wrong now

exploitagency, 2018-09-17:
That sketch is outdated. Run the latest software without modification and see if it works, then try changing values again.

https://github.com/exploitagency/ESPloitV2

PaulSec, 2018-09-17:
Hey there,

Thanks for the inputs. I will give this a shot tomorrow and let you know how it goes.

whid-injector, 2018-09-17:
Yeah, Corey is right! 
U have to try the ESPLOITV2 sketch, since ESPLOITV2  is the fw shipped by default nowadays!

https://github.com/exploitagency/ESPloitV2/blob/master/source/Arduino_32u4_Code/Arduino_32u4_Code.ino

PaulSec, 2018-09-18:
Hey there, 

After uploading the new sketch, this works like a charm! Thanks a lot for your help :beers:

closed 2018-09-18 by PaulSec