whid-injector/WHID #29: Keyboard layout PT-BR

stephanodesouza, 2018-10-30:
I'm new to Arduino. I need to change my keyboard layout, I've downloaded PT-BR, but how do I connect my CACTUS WHID to the Ardino IDE and perform the upgrade? I did not find any beginner tutorial. Thank you

stephanodesouza, 2018-10-31:
I would be able to teach, for lay users like me, how to connect Cidus Whid in the Arduino IDE, to change the layout of the keyboard. In your tutorial teaches you to only modify the Keyborad.cpp file, but does not talk about how to upload the modicication through the IDE. Not everyone who bought Cactus Whid has a high knowledge of arduino. Thank you

whid-injector, 2018-11-01:
https://github.com/exploitagency/ESPloitV2/blob/36638cfe1a4fb3749355faf63ef182c19419dd52/source/Arduino_32u4_Code/Arduino_32u4_Code.ino

P.S. If you cannot understand such obvious thing, DO NOT play with hacking toys.

closed 2018-11-01 by whid-injector