whid-injector/WHID #30: Whid stuck in auto payload mode

lazyjay559, 2018-11-30:
So I’m trying to get back into the menu but can’t because it keeps typing. I can’t get in to change it back to not launch the payload when it’s plugged in. Tried to reset software following directions with IDE and the ESP tool but the esp tool doesn’t connect even tho it’s plugged in. It still runs the payload though non stop. I tried the hard reset too but didn’t do anything. Any ideas? Thanks in advance.

exploitagency, 2018-11-30:
Plug it into a usb charger and not into a machine.

lazyjay559, 2018-11-30:
I did that, I can’t connect to the ip even after connecting to correct wifi

lazyjay559, 2018-11-30:
I run ifconfig and see that ip is 192.168.1.1 and it’s juat not connecting

closed 2018-12-02 by lazyjay559