whid-injector/WHID #31: Can’t connect to Ip because auto payload

lazyjay559, 2018-12-02:
I have auto payload selected. I wanted to change it back but now I cannot access the control page. I’m doing everything right connect to exploit and the correct ip but the page does not load and my whid keeps running script non stop until it’s unplugged. It doesn’t just run a few times and done. I also tried a hard reset with magnet but that won’t work either due to it continually running the script. I first plugged into an external power bank. The “exploit” ap shows and connects. I type in whatever the address was (can’t think what it is off the top of my head) and it just doesn’t load. Ive tried everything and no lick this is the 3rd day and I’m getting pissed now. If someone could help that would be amazing.

lazyjay559, 2018-12-05:
Any help here?????

exploitagency, 2018-12-06:
Tap a ground on the reset pin or use a magnet on the reed switch to reset the device while avrdude is trying to flash it before the payload starts.  It will work if you time things right.

exploitagency, 2018-12-06:
Tap a ground on the reset pin of the 32u4 while trying to flash it and avrdude is looking for the device.  It will take good timing and a few tries.