whid-injector/WHID #35: It IS not showing in my Wifi Options

InfiniteScope, 2019-01-29:
Hello I tried Resetting to default configuration/Recovering device but it didn`t work can tou help, I can`t connect to the injector I changed the username and the password and it was gone!
When I try to reset it is saying:
Relaying command to connected ESP device.
Please help I just bought the USB!
Thanks!

InfiniteScope, 2019-01-30:
@whid-injector

Grave-Digger, 2019-02-22:
Also having this issue after doing same thing. The SSID changed to AI-Thinker *series of numbers here*

Letomg0, 2019-08-05:
> Also having this issue after doing same thing. The SSID changed to AI-Thinker _series of numbers here_
> @whid-injector

Did you ever fix this? I've run though the 4 stage re-flash tutorial and have the same issue - the device is making an access point called 'AI-THINKER-323FDC' but when I connect to it can't reach 192.168.1.1 etc.

whid-injector, 2019-08-06:
Was the whid working when u received it? Yes.
You fucked it up? Most likely.
The wiki and documentation is clear enough to fix issues. Try harder.

closed 2019-08-06 by whid-injector