whid-injector/WHID #37: Settings change causes bricked device.

dwhacks, 2019-02-28:
Which OS are you using: Linux Mint
    Which Arduino IDE Version: 1.8.2, 1.8.5, 1.8.8 1.6.5
    The ESPtool you are using is from https://github.com/whid-injector/esptool: Shouldnt matter, not using it
    The USB port you are using is USB 3.0: tried powered hub and 2.0 port
    Did you try with an older version of Arduino IDE (i.e. 1.6.0) 1.6.5
    Did you try to use an USB 2.0 port: yes
    Did you try with another laptop/PC: now

So I just got my two usb dongles, when I plug them in it does what it says and seems to work. I went into the settings to change the password on the first one and then when I clicked to save, it disapeared from the network and then the LED just blinks contantly. I tried pasting in the reset into the arduino serial monitor and it replies with "Relaying command to connected ESP device." but nothing seems to change.

leocelente, 2019-03-06:
I had a similar problem after turning FTP on. I unbricked it following [this](https://github.com/exploitagency/ESPloitV2/tree/36638cfe1a4fb3749355faf63ef182c19419dd52/flashing) and [this](https://www.youtube.com/watch?v=MRGUSPW-Cr0). 

Use only files contained in this repo and the linked version of ESPloiV2. 
Be aware of versions and the settings in the flasher. 
(I had to do it twice because I forgot to set the flasher's baudrate correctly).

Hope it works!

Seanmeister, 2019-03-07:
I have went through twice on a windows machine following the links that leocelente posted and triple checked everything along the way.

Everything goes through correctly but the device led is just flashing randomly and nothing else no wifi or anything.

Any ideas?

dwhacks, 2019-03-07:
I managed to get mine going, but i'm on linux so that might be a bit different. 

Also, I had to run my arduino environment as Root or it would not flash the arduino portion.

So first flash the arduino portion with the "esp flasher" sketch, as per the recovery steps.

Then I had to get esptool from this repository, and then flash the esp from its directory using this:
`./esptool.py -p /dev/ttyACM0 --baud 115000 write_flash 0x00000 /home/dwhacks/ESPloitV2-master/source/ESP_Code/ESP_generic.bin --flash_size 32m`

Before running this I tested to make sure esptool could see it with:
`./esptool.py -p /dev/ttyACM0 flash_id`
But after doing this I would have to unplug as replug the WHID verifying it was seen using the LSUSB command.

Seanmeister, 2019-03-07:
Thank you @dwhacks I am going to try to do it on a linux setup

ShutdownRepo, 2020-01-11:
I've had the same issue here by changing the delay times using the WHID Connector app.
I have tried the whole inital flashing and hardware unbricks without luck. Did anyone manage to unbrick the dongle ?

TheHyp3ri0n, 2020-01-11:
No luck with mine. I ended up just buying a new one.

fishso, 2020-08-04:
1.read https://github.com/whid-injector/WHID/wiki 
2.read https://github.com/whid-injector/WHID/tree/master/ESPloitV2_whid,  try `Initial Flashing`, I have done it successfully just now。