whid-injector/WHID #41: WHID stopped working

Seanmeister, 2019-03-08:
I received my WHID Cactus yesterday. I plugged it into my computer and all was working. I tried to make a change in settings and clicked save. The wifi disappeared and I waited 10 mins still no change. I unplugged the device and plugged it back in. The led started flashing repeatedly. I have been through reflashing it several times and everything says its good. Now when you plug it in the led flashes every min or so. Am I screwed?

Which OS are you using?
I have tried 2 different computers running Windows and 1 running Linux

Which Arduino IDE Version?
I have tried 3 different version

The ESPtool you are using is from https://github.com/whid-injector/esptool ?
I tried it from that link as well as nodemcu-flasher on windows

The USB port you are using is USB 3.0? (sometimes they may create issues with arduino boards. try a usb 2.0 or a usb hub)
I have tried 2.0 and 3.0

Did you try with an older version of Arduino IDE (i.e. 1.6.0)?
Yes

Did you try to use an USB 2.0 port?
Yes

Did you try with another laptop/PC?
Yes

FedeCuci, 2021-03-28:
Having the same issue and tried everything you have said @theman23290. Unfortunately, it still doesn't seem to work. Have you figured anything out @TheHyp3ri0n ?

TheHyp3ri0n, 2021-03-29:
No I ended up buying a new one. Sorry.

FrancisLaTruelle, 2021-06-12:
Did you find a solution please @FedeCuci ?

FedeCuci, 2021-06-12:
No, sorry @FrancisLaTruelle.