whid-injector/WHID #42: After I changed the SSID and password, SSID became invisible and blue light started blinking

vcchm, 2019-03-08:
The device came pre-installed. It worked correctly until I changed the SSID and password. The the SSID became invisible and blue light started blinking and I could not use it anymore.

I tried reflashing it using the procedure proposed here : https://github.com/exploitagency/ESPloitV2
I had no error message installing the loader, the ESP flashed correctly as well as the third code (no error message) but the device is still not working
.
I tried erasing the internal memory by flashing a 4MB blank file before flashing the ESP code. It does not work either. I tried installing a release binary (ie not compiling the ESP code myself) and it does not work either.

What did I do wrong ? Is there a bug in the code when you change the SSID name ?

vcchm, 2019-03-08:
Ok, I flashed it using NodeMcuFirmware Flasher and 57600 bauds speed, and it worked.
IMHA the only things that need to be fixed is
1 - mentioning the limited speed in the flashing manual, and maybe the need the clear the flahs when re-installing
2- fixing the bug that bricks the WHID when changing the settings. What I have done is changing the arduino code to choose a different default name.

ENCRYPTEDGH057, 2019-03-14:
I am facing same problem :(

unbaiat, 2019-03-14:
same problem here

unbaiat, 2019-03-14:
@vcchm thanks! it worked!

aruso, 2019-03-20:
@vcchm @unbaiat can you explain how you solved this? I just got 2 devices ... changed the ssid and pass and both lost the SSID. I don't see any ssid for these anymore. You mention re-flashing ... if so what file do I re-flash and how exactly. Thanks

jfrmilner, 2019-04-08:
@aruso If you've locked yourself out with an invalid network configuration then first try a soft config reset over serial [Software Unbrick & Reset to Default Configuration](https://github.com/whid-injector/WHID/wiki#software-unbrick--reset-to-default-configuration)

aruso, 2019-04-08:
Tried that but it didn't work ... And neither does flashing ... I have 2
devices like this... Both stopped working after a basic change and now are
useless. Very flaky.

whid-injector, 2019-08-06:
the hard reset feature is clear. if you are not able to flash a fw... don't even buy an opensource hacking device.

closed 2019-08-06 by whid-injector