whid-injector/WHID #43: Device not recognized - Not being able to perform a Hard-Reset

demian123, 2019-04-18:
`
    Which OS are you using? Windows 7 and Kali Linux VM
    Which Arduino IDE Version? 1.8.9
    The ESPtool you are using is from https://github.com/whid-injector/esptool ? Yes
    The USB port you are using is USB 3.0? (sometimes they may create issues with arduino boards. try a usb 2.0 or a usb hub) Tried both 2.0 and 3.0
    Did you try with an older version of Arduino IDE (i.e. 1.6.0)? No
    Did you try to use an USB 2.0 port? Yes
    Did you try with another laptop/PC? Yes
`
Hey guys, today i received my Cactus WHID, i had the famous problem involving not being able to properly see the AP because i changed the SSID.

So i followed this guide: https://github.com/exploitagency/ESPloitV2/tree/36638cfe1a4fb3749355faf63ef182c19419dd52 , made a mistake in one of the steps, and now no matter in wich device i insert the Cactus WHID, i get the "Unknown Device" thing ("the USB device is not recognized").

Should i try a hard-reset (with the magnet and all)? I tried but it's kind of complicated to accomplish, plus the tutorial in the Wiki it's not that friendly: https://github.com/whid-injector/WHID/wiki#hardware-unbrick--hard-reset

EDIT: Already tried with three different tipes of magnets and nothing seems to happen, is there any special type of magnet that i need to use?

Thanks in advance for any tips!

demian123, 2019-04-29:
Anyone..?

whid-injector, 2019-08-06:
try harder.

closed 2019-08-06 by whid-injector

tonysader, 2022-01-26:
Hey bro,
any advice? I changed the SSID too, 
then I didn't have any access point.

I tried the magnet, I tried to flashing it with the sketches like the tutorial but still have no results.

Thanks 
@whid-injector  @demian123