whid-injector/WHID #45: Possible to inject payloads when target connects to AP?

clearlyunseen, 2019-07-09:
Im looking for a way to set up the cactus as an AP that runs a script on any device that connects to it. Is this possible?

kratos13, 2019-07-10:
I mean... it is an AP, that runs DHCP... so, in theory, you could write a script that recognizes when an IP address has been handed out, and then execute "x" script. I haven't played with my cactus in a while (the payload size is too small to run what I need), but still a really fun tool for POC's.

krakrukra, 2019-07-13:
No, it's not possible. WHID is basically a remotely controlled keyboard, plugged into a particular machine. That means you can not type into any other machine, since WiFi routers (access points) do not send keystrokes over network.

closed 2019-08-06 by whid-injector