whid-injector/WHID #46: Help with language da-DK

mitmix, 2019-07-15:
Hello,

i got hold if a cactus WHID, but i seem to have a problem with the language packs available,
i downloaded the layouts and tried the automated solution as well,  i have tried to replace the ASCIImap section with the one from the language file with no luck and tried the automated solution by copying the whole file and this solution didnt work either, i can get it running but all chars requiring ALTgr fail such as @, $.

my approach is as follows,
update the keyboard.cpp file with the danish ASCIImap
Flash the lilly
compile ESP_code 
upload the arduino_32u4_Code 

am i missing something ?

this is the part copied/replaced in to the keyboard.cpp with the excisting #define 
#define SHIFT 0x80
const uint8_t _asciimap[128] =
{
  0x00,             // NUL
  0x00,             // SOH
  0x00,             // STX
  0x00,             // ETX
  0x00,             // EOT
  0x00,             // ENQ
  0x00,             // ACK
  0x00,             // BEL
  0x2a,      // BS  Backspace
  0x2b,      // TAB  Tab
  0x28,      // LF  Enter
  0x00,             // VT
  0x00,             // FF
  0x00,             // CR
  0x00,             // SO
  0x00,             // SI
  0x00,             // DEL
  0x00,             // DC1
  0x00,             // DC2
  0x00,             // DC3
  0x00,             // DC4
  0x00,             // NAK
  0x00,             // SYN
  0x00,             // ETB
  0x00,             // CAN
  0x00,             // EM
  0x00,             // SUB
  0x00,             // ESC
  0x00,             // FS
  0x00,             // GS
  0x00,             // RS
  0x00,             // US

  0x2c,       //  ' '
  0x1e|SHIFT,     // !
  0x1f|SHIFT,     // "
  0x20|SHIFT,    // #
  0x21,          // $
  0x22|SHIFT,    // %
  0x23|SHIFT,    // &
  0x31,          // '
  0x25|SHIFT,    // (
  0x26|SHIFT,    // )
  0x31|SHIFT,    // *
  0x2d,          // +
  0x36,          // ,
  0x38,          // -
  0x37,          // .
  0x24|SHIFT,    // /
  0x27,          // 0
  0x1e,          // 1
  0x1f,          // 2
  0x20,          // 3
  0x21,          // 4
  0x22,          // 5
  0x23,          // 6
  0x24,          // 7
  0x25,          // 8
  0x26,          // 9
  0x37|SHIFT,    // :
  0x36|SHIFT,    // ;
  0x64,          // <
  0x27|SHIFT,      // =
  0x64|SHIFT,      // >
  0x2d|SHIFT,      // ?
  0x1f,            // @
  0x04|SHIFT,      // A
  0x05|SHIFT,      // B
  0x06|SHIFT,      // C
  0x07|SHIFT,      // D
  0x08|SHIFT,      // E
  0x09|SHIFT,      // F
  0x0a|SHIFT,      // G
  0x0b|SHIFT,      // H
  0x0c|SHIFT,      // I
  0x0d|SHIFT,      // J
  0x0e|SHIFT,      // K
  0x0f|SHIFT,      // L
  0x10|SHIFT,      // M
  0x11|SHIFT,      // N
  0x12|SHIFT,      // O
  0x13|SHIFT,      // P
  0x14|SHIFT,      // Q
  0x15|SHIFT,      // R
  0x16|SHIFT,      // S
  0x17|SHIFT,      // T
  0x18|SHIFT,      // U
  0x19|SHIFT,      // V
  0x1a|SHIFT,      // W
  0x1b|SHIFT,      // X
  0x1c|SHIFT,      // Y
  0x1d|SHIFT,      // Z
  0x25,          // [
  0x64,          // bslash
  0x26,          // ]
  0x30|SHIFT,    // ^
  0x38|SHIFT,    // _
  0x2e|SHIFT,    // `
  0x04,          // a
  0x05,          // b
  0x06,          // c
  0x07,          // d
  0x08,          // e
  0x09,          // f
  0x0a,          // g
  0x0b,          // h
  0x0c,          // i
  0x0d,          // j
  0x0e,          // k
  0x0f,          // l
  0x10,          // m
  0x11,          // n
  0x12,          // o
  0x13,          // p
  0x14,          // q
  0x15,          // r
  0x16,          // s
  0x17,          // t
  0x18,          // u
  0x19,          // v
  0x1a,          // w
  0x1b,          // x
  0x1c,          // y
  0x1d,          // z
  0x24,          // {
  0x2e,          // |
  0x27,          // }
  0x30,          // ~
  0        // DEL
};

// Init var
bool _altGrMap[128];
bool _altFine = false;

// Individually define all needed char
void initAltGr() {
  _altFine = true;

  _altGrMap[36] = true; // $
  _altGrMap[64] = true; // @
  _altGrMap[91] = true; // [
  _altGrMap[92] = true; // bslash
  _altGrMap[93] = true; // ]
  _altGrMap[123] = true; // {
  _altGrMap[124] = true; // |
  _altGrMap[125] = true; // }
  _altGrMap[126] = true; // ~
}

Thanks in advance

mitmix, 2019-07-16:
if (!_altFine) initAltGr();

    if (_altGrMap[oldKey]){
      _keyReport.modifiers |= 0x40;
    } else {
      _keyReport.modifiers = 0;
    }

mitmix, 2019-07-16:
Added the bold text and now it can press ALTgr but now it wont release it again after press 

size_t Keyboard_::press(uint8_t k) 
{
	uint8_t i;
	if (k >= 136) {			// it's a non-printing key (not a modifier)
		k = k - 136;
	} else if (k >= 128) {	// it's a modifier key
		_keyReport.modifiers |= (1<<(k-128));
		k = 0;
	} else {		// it's a printing key
		**int oldKey = k;**
		k = pgm_read_byte(_asciimap + k);
		if (!k) {
			setWriteError();
			return 0;
		}
		
		
		
		 **if (!_altFine) initAltGr();

    if (_altGrMap[oldKey]){
      _keyReport.modifiers |= 0x40;
    } else {
      _keyReport.modifiers = 0;
    }**
		
		
		
		if (k & 0x80) {						// it's a capital letter or other character reached with shift
			_keyReport.modifiers |= 0x02;	// the left shift modifier
			k &= 0x7F;
		}
	}

danielkp1234, 2019-08-14:
Have u got it working?

leona, 2020-08-05:
Bit late on this but I was having a similar issue and found a solution. I used LocaleKeyboard.BAT to generate the .cpp and replaced that with the Arduino one. Didn't modify any files. Only thing was I had to use Arduino IDE 1.8.1 any earlier and it wouldn't compile for a million reasons and any later and it complained about Keyboard.h