whid-injector/WHID #69: Cactus WHID not working!!!!

JustAnass, 2021-01-09:
30 minutes after I got my Cactus Whid I got to work, I had changed a few settings: (IP) & "Use Existing Network" not knowing what it entailed. After I did that the network "exploit" disappeared and only my home wifi was in the network list. Resetting also doesn't work when I use the "ResetDefaultConfig:" in Arduino IDE, nothing happens, no light, no output.

electricWah, 2021-01-11:
Check the COM port and the baud rate, and make sure you send it exactly as written.
I did the exact same thing in the first 10 mins ;)

JustAnass, 2021-01-12:
> Check the COM port and the baud rate, and make sure you send it exactly as written.
> I did the exact same thing in the first 10 mins ;)

The COM port was first COM 7 and now it stays on COM 8? And what is an baud rate.
Can you tell me what you did to fix it?

electricWah, 2021-01-13:
The baud rate is the data transfer speed, and is required in order to send/receive data through serial. The fact you do not know what it is suggests what the problem is ;)
Go to the wiki [here](https://github.com/whid-injector/WHID/tree/master/ESPloitV2_whid#resetting-to-default-configurationrecovering-device) and follow the instructions exactly, they are very detailed. Good luck!

JustAnass, 2021-01-13:
> The baud rate is the data transfer speed, and is required in order to send/receive data through serial. The fact you do not know what it is suggests what the problem is ;)
> Go to the wiki [...]

I already followed the "esetting to default configuration/Recovering device" chapter and chapter "Initial Flashing", but it didnt work. I even asked help in a dc server and they said they didnt either why it was not working

GamingMetJayden, 2021-01-21:
I already followed the "esetting to default configuration/Recovering device" chapter and chapter "Initial Flashing", but it didnt work. I even asked help in a dc server and they said they didnt either why it was not working

I did exactly the same.. Now it won't work like yours! I dont have a flashing light when I plug it in.. So I think I ruined my USB... Any people that now how to solve this?

tonysader, 2022-01-26:
The same problem here,

Did you manage to solve it ?
@GamingMetJayden  @gggrafff  @whid-injector

gggrafff, 2022-05-04:
My problem is here: https://github.com/whid-injector/WHID/issues/57
Check please that you have the same problem.

It helped me...
Original (In Russian): http://wiki.amperka.ru/установка-и-настройка-arduino-ide:arduino-restore-bootloader
Translation (In English): https://wiki-amperka-ru.translate.goog/установка-и-настройка-arduino-ide:arduino-restore-bootloader?_x_tr_sch=http&_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en