whid-injector/WHID #73: New Wifi Duck compatibility ?

Hamidw, 2021-04-04:
Hey, i recently bought the WHID Cactus and i really like it, however i got into all settings and did everything i can do with ESPloitV2 ans want to spice it up by changing firmware, that's why im asking if the ameliorated wifi-ducky by spacehuhn (Wifi Duck) is supporterd because i don't see it on the wiki. And if it's compatible, how do i switch to it from ESPloit. Thank you very much for this amazing project and keep going ! Greetings from a 14yo french computer enthusiast💪🏼

IMExperts, 2021-04-09:
I'm also looking forward to this..

TQMatvey, 2021-05-29:
better dont change it i tried reflashing with esploitv2 again and now i have ai-thinker that doesnt work