whid-injector/WHID #77: EsploitV2 flashing s#it

TQMatvey, 2021-05-29:
Hello, yesterday i was trying to reflash my whid cactus cuz yes, so i did everything as in tutorial, and at the end my usb cactus is now open network named AI-THINKER-<somerandomshit>
so wtf i reflashed again same... i am tired i just wasted like 25$ on this and it doesnt work and the devs left the project and almost noone here helps :c even tho the network is open but when i connect it still doesnt let me to 192.168.1.1 and even the def gateway is 192.168.4.1 which is mostly used in spacehuhn's projects BUT EVEN THAT GATEWAY LEADS ME TO NOTHING ERROR LOADING. what do i do now i tried everything, hard resets, reflashes NOTHING i am in bad mood third day as my whid cactus is useless now

TQMatvey, 2021-05-30:
ok i were able to reflash using older compatible version which is stupid but whatevernow i wanna ask why the fuk auto script doesnt work, i have set it to my rick roll and it only works on reboot not on injection

FrancisLaTruelle, 2021-06-11:
Hello, I am in the same case, which version do you use for flashing and have you made changes compared to the tutorial please?

closed 2021-11-26 by TQMatvey

nikoli0218, 2023-05-16:
Did any1 figure this out ...i had forgotten what i set the ap password to so figured id just reflash and ya ..Ai thinker ?? Is there a way to restore this to factory defaults after a flash?