whid-injector/WHID #83: A couple issues (yes I read the wiki)

GabrielSestieri, 2021-09-30:
I bought a cactus that came pre installed with everything. Like many of the people here I tried changing the name and password of the device and the next time I plugged it back in it no longer showed up. I am on a Mac version 10.15.7  and the latest version of Arduino.I read every line in the wiki and scoured all the issues both opened and closed. I tried rebooting both software and hardware to no avail. The problems I ran into are as follows:
1. Re-flashing it would generate a "ESP8266FtpServer.h: No such file or directory" I tried with different versions of both the ESP8266 library and different versions of the Json library (since the wiki has been written new versions have been released so I used the exact ones in the wiki and vice versa). Once I realized I don't necessarily need to re-flash I went on to:

2. Software Unbricking did not show me a "Resetting configuration files back to default settings." It just didn't do anything, just cleared the input field. 

3. Hardware Unbricking, which is the one I thought I'd have the most luck with since it's rather simple steps was the most annoying. For the life of me I could not get the bootloader port to show up. I ran a magnet under the usb like mentioned (close-away-close-away, like in the YouTube video, two blue flashes) and quickly went to Tools > Ports but the only port showing up was: 
"dev.cu.usbmodem1412101 (LilyPad Arduino USB)" no bootloader in sight. 
Regardless I still tried the mentioned steps and still has not showed up in my WIFI.

Any help would be gladly appreciated I've been at this for like 3 and half hours. 

BTW the USB flashes every time I plug it in and flashes every time I run a magnet under it.

HugoMskn, 2022-03-05:
Hey, contact me on discord, I'll try to help you ! 
Hugo ;)#4872

GabrielSestieri, 2022-03-05:
> Hey, contact me on discord, I'll try to help you ! 
> 
> Hugo ;)#4872

Hey man I couldn't find you :/ add me instead 

gabsec#0103