whid-injector/WHID #87: Need Help! "No internet connection"

SmallShard, 2022-02-03:
This message keeps coming up in my wifi settings when I connect to the wifi. I just started using the cactus and it was working fine. I come back 20 minutes later and this message keeps coming up. I haven't changed many settings in the esploit config. [image]

HugoMskn, 2022-03-04:
that's normal, I dont think there's a way to get the host wifi with the cactus whid, I just received it so i'll tel you

SmallShard, 2022-03-08:
> received 

Correction, I was not using a Linux device, however this issue still occurs on windows 10 so I don't think the operating system is involved

dedseci0i, 2022-04-02:
it only goes to the ip page, 192.168.1.1
try going there when connected to exploit and it should be fine