whid-injector/WHID #91: i just got my whid cactus and when i plug it in i get ap called wireless-usb-disk

vokica, 2022-04-02:
my cactus just arrived from aliexpress and when i connect to it its not loaded with esploit idk what to do?

dedseci0i, 2022-04-07:
Hey! Can you send a link to the buying link via AliExpress?

HankDaHacker, 2022-04-19:
I have the same issue. Right out of the box mine has an access point of wireless-usb-disk..
No password was given.  No light on the usb stick when plugged in.  I've tried it on 5 different computers and laptops with no access point called exploit.  This thing garbage or what?

dedseci0i, 2022-04-19:
I would go to Arduino and reflash it

xWvdHx, 2022-04-27:
@HankDaHacker your default password is "12345678" I have the same problem.I ordered one from Amazon, and checked all the product info before ordering and it told me it really was an WHID cactus USB. But only thing I see is the Wireless-USB-disk when plugging the USB in to my laptop and it doesn't even has internet when you connect to the WiFi. 

I first thought they made a mistake and sent the wrong product but got the same problem after sending the USB back and buying a  new one.

vokica, 2022-04-28:
> Hey! Can you send a link to the buying link via AliExpress?

https://a.aliexpress.com/_mtoOS5C

vokica, 2022-06-26:
> I'm having the same issue and what I'm finding is that our usb'a are flashed with a demo code WUD. I've been trying to figure out a way to flash whid cactus to it I can't figure it out if anyone figured this out please let me know

i tried  with arduino but when i select only port com1  it says no device on that port itried googling but still nothing

Sewpee, 2022-06-27:
> i tried  with arduino but when i select only port com1  it says no device on that port itried googling but still nothing

Yeah same for me. I just ended up buying an O.MG Plug from hak5

whid-injector, 2023-09-05:
Probably you bought WUD and NOT WHID!
The correct Wiki for WUD is https://wiki.aprbrother.com/en/wud.html

closed 2023-09-05 by whid-injector

dedseci0i, 2023-09-08:
u think im retarded? i have your app with the sativa leaf off the kali
nethunter store and everything. i have the whid