whid-injector/WHID #99: bro why r u scamming

Adamladeb, 2023-02-14:
i just bought this dumb shit with 24$
and it's not working

ihsaank, 2023-09-08:
Bro I’ve faced this same problem aswell when I try to reflash:

exit status 1

Compilation error: no return statement in function returning non-void [-Werror=return-type]

Have you found the solution yet?

whid-injector, 2023-09-08:
- First and foremost... I AM NOT YOUR BRO.
# How to report bugs:
I won't spend time explaining in depth how to report issues, since I am pretty sure you have done it tons of times. 
The main idea can be summed up in the following points:
1. Expected Behavior
2. Unexpected Behavior
3. which OS you using?
4. Which Arduino IDE version you using?
5. Is the target computer an Apple Product?
6. Is the target USB port an USB3.0?
7. Eventual Explanation Notes, Screenshots, Videos, etc.

Adamladeb, 2023-09-08:
yea i fixed it i don't remeber how but i forgot a step u should watch the
vid tutorial